dlktdr/BTWifiModule #2: Flashing went wrong

yammaster, 2022-06-22:
I use this board:

[image]

I flash as ESP32 files from Node32s_Wroom32. Flashing finished, but in serial monitor I see this messages:

[image]

Whats wrong?

dlktdr, 2022-06-22:
Looks like it either didn't flash properly or maybe to the wrong locations?

That check box in the flashing config for donotchgbin can you try it in the opposite. I had a problem flashing myself using that method when I unchecked it worked.

Maybe send the log file here too

yammaster, 2022-06-22:
`*************************** START ****************************

START TIME: 202206222156

CONNECT BAUD: 115200
set state: ESP_DL_SYNC
serial port opened
-----------
baud:115200
root baud:115200
-------------
===============BAUD : 115200===============CALL DEVICE SYNC
connecting...
chip sync ok
mac l: 0x6d10ec34
mac h: 0xd78e3
crc_cal: 13
crc_read: 13
crc_test: 0
ESP32 MAC CRC OK
get mac res: 1
get flash id : 0x0016405e

manufacturer_id: 0x5e

device_id: 0x4016

Wrote 26256 bytes (16027 compressed) at 0x00000000 in 0.2 seconds (effective 970.1 kbit/s)...

Hash of data verified.
Wrote 3072 bytes (103 compressed) at 0x00008000 in 0.0 seconds (effective 3161.6 kbit/s)...

Hash of data verified.
Wrote 872000 bytes (496244 compressed) at 0x00010000 in 8.7 seconds (effective 799.8 kbit/s)...

Hash of data verified.
Leaving...

com closed
Use time: 11 s

Download result: PASS

*************************** END ****************************
`

yammaster, 2022-06-22:
And this is the result with the esphome-flasher:

`ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:8896
load:0x40080400,len:5828
entry 0x400806ac
I (357) cpu_start: Pro cpu up.
I (357) cpu_start: Starting app cpu, entry point is 0x40081ad8
I (0) cpu_start: App cpu up.
D (365) clk: RTC_SLOW_CLK calibration value: 3552922
I (376) cpu_start: Pro cpu start user code
I (376) cpu_start: cpu freq: 240000000
I (376) cpu_start: Application information:
I (381) cpu_start: Project name:     BTWifiModule
I (386) cpu_start: App version:      ed8f590
I (391) cpu_start: Compile time:     Jun 20 2022 00:35:57
I (397) cpu_start: ELF file SHA256:  2c9216916e141503...
I (403) cpu_start: ESP-IDF:          4.3.2
D (408) memory_layout: Checking 11 reserved memory ranges:
D (413) memory_layout: Reserved memory range 0x3ffae000 - 0x3ffae6e0
D (420) memory_layout: Reserved memory range 0x3ffae6e0 - 0x3ffaff10
D (426) memory_layout: Reserved memory range 0x3ffb0000 - 0x3ffb6388
D (433) memory_layout: Reserved memory range 0x3ffb8000 - 0x3ffb9a20
D (439) memory_layout: Reserved memory range 0x3ffbdb28 - 0x3ffbdb5c
D (445) memory_layout: Reserved memory range 0x3ffbdb60 - 0x3ffc6498
D (452) memory_layout: Reserved memory range 0x3ffe0000 - 0x3ffe0440
D (458) memory_layout: Reserved memory range 0x3ffe3f20 - 0x3ffe4350
D (465) memory_layout: Reserved memory range 0x40070000 - 0x40078000
D (471) memory_layout: Reserved memory range 0x40078000 - 0x40080000
D (478) memory_layout: Reserved memory range 0x40080000 - 0x40097fac
D (484) memory_layout: Building list of available memory regions:
D (490) memory_layout: Available memory region 0x3ffaff10 - 0x3ffb0000
D (497) memory_layout: Available memory region 0x3ffb6388 - 0x3ffb8000
D (503) memory_layout: Available memory region 0x3ffb9a20 - 0x3ffbdb28
D (510) memory_layout: Available memory region 0x3ffc6498 - 0x3ffc8000
D (517) memory_layout: Available memory region 0x3ffc8000 - 0x3ffca000
D (523) memory_layout: Available memory region 0x3ffca000 - 0x3ffcc000
D (530) memory_layout: Available memory region 0x3ffcc000 - 0x3ffce000
D (536) memory_layout: Available memory region 0x3ffce000 - 0x3ffd0000
D (543) memory_layout: Available memory region 0x3ffd0000 - 0x3ffd2000
D (550) memory_layout: Available memory region 0x3ffd2000 - 0x3ffd4000
D (556) memory_layout: Available memory region 0x3ffd4000 - 0x3ffd6000
D (563) memory_layout: Available memory region 0x3ffd6000 - 0x3ffd8000
D (569) memory_layout: Available memory region 0x3ffd8000 - 0x3ffda000
D (576) memory_layout: Available memory region 0x3ffda000 - 0x3ffdc000
D (583) memory_layout: Available memory region 0x3ffdc000 - 0x3ffde000
D (589) memory_layout: Available memory region 0x3ffde000 - 0x3ffe0000
D (596) memory_layout: Available memory region 0x3ffe0440 - 0x3ffe3f20
D (602) memory_layout: Available memory region 0x3ffe4350 - 0x3ffe8000
D (609) memory_layout: Available memory region 0x3ffe8000 - 0x3fff0000
D (616) memory_layout: Available memory region 0x3fff0000 - 0x3fff8000
D (622) memory_layout: Available memory region 0x3fff8000 - 0x3fffc000
D (629) memory_layout: Available memory region 0x3fffc000 - 0x40000000
D (635) memory_layout: Available memory region 0x40097fac - 0x40098000
D (642) memory_layout: Available memory region 0x40098000 - 0x4009a000
D (649) memory_layout: Available memory region 0x4009a000 - 0x4009c000
D (655) memory_layout: Available memory region 0x4009c000 - 0x4009e000
D (662) memory_layout: Available memory region 0x4009e000 - 0x400a0000
I (668) heap_init: Initializing. RAM available for dynamic allocation:
I (676) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
D (682) heap_init: New heap initialised at 0x3ffb6388
I (687) heap_init: At 3FFB6388 len 00001C78 (7 KiB): DRAM
D (693) heap_init: New heap initialised at 0x3ffb9a20
[... 258 lines cut ...]
D (1616) efuse: In EFUSE_BLK0__DATA1_REG is used 8 bits starting with 24 bit
D (1626) efuse: coding scheme 0
D (1626) efuse: In EFUSE_BLK0__DATA1_REG is used 8 bits starting with 16 bit
D (1636) efuse: coding scheme 0
D (1636) efuse: In EFUSE_BLK0__DATA1_REG is used 8 bits starting with 8 bit
D (1646) efuse: coding scheme 0
D (1646) efuse: In EFUSE_BLK0__DATA1_REG is used 8 bits starting with 0 bit
D (1656) efuse: coding scheme 0
D (1656) efuse: In EFUSE_BLK0__DATA2_REG is used 8 bits starting with 16 bit
D (1916) nvs: nvs_open_from_partition bt_config.conf 1
D (1916) nvs: nvs_get_str_or_blob bt_cfg_key0
D (1916) nvs: nvs_get_str_or_blob bt_cfg_key0
D (1916) nvs: nvs_close 3
D (1916) nvs: nvs_open_from_partition bt_config.conf 1
D (1926) nvs: nvs_set_blob bt_cfg_key0 216
D (1926) nvs: nvs_close 4
I (1996) BTCLIENT: Starting Central
I (1996) BTCLIENT: REG_EVT
Connecting to 000000000000
I (2506) SETTINGS: Reading settings
D (2506) nvs: nvs_get_str_or_blob skey
I (2506) SETTINGS: Settings Read Successfully
I (2506) SETTINGS: No data differs, not writing to flash

`

dlktdr, 2022-06-22:
Well from that one it appears that the program is actually written on there and running. I can see some of the things I wrote in the log. 

But it looks like you have a power problem, there's a brown out detection listed in there and then it reboots.

Are you still powered from USB or is this powered from the radio?

yammaster, 2022-06-23:
The board was powerd by the pc.
I used another board, flashed with the new firmware and it looks like better in my opinion:

`ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:8896
load:0x40080400,len:5828
entry 0x400806ac
I (357) cpu_start: Pro cpu up.
I (357) cpu_start: Starting app cpu, entry point is 0x40081ad8
I (0) cpu_start: App cpu up.
D (365) clk: RTC_SLOW_CLK calibration value: 3570330
I (376) cpu_start: Pro cpu start user code
I (376) cpu_start: cpu freq: 240000000
I (376) cpu_start: Application information:
I (381) cpu_start: Project name:     BTWifiModule
I (386) cpu_start: App version:      f6c66af
I (391) cpu_start: Compile time:     Jun 23 2022 03:14:39
I (397) cpu_start: ELF file SHA256:  ac13ed1585333605...
I (403) cpu_start: ESP-IDF:          4.3.2
D (408) memory_layout: Checking 11 reserved memory ranges:
D (413) memory_layout: Reserved memory range 0x3ffae000 - 0x3ffae6e0
D (420) memory_layout: Reserved memory range 0x3ffae6e0 - 0x3ffaff10
D (426) memory_layout: Reserved memory range 0x3ffb0000 - 0x3ffb6388
D (433) memory_layout: Reserved memory range 0x3ffb8000 - 0x3ffb9a20
D (439) memory_layout: Reserved memory range 0x3ffbdb28 - 0x3ffbdb5c
D (445) memory_layout: Reserved memory range 0x3ffbdb60 - 0x3ffc6498
D (452) memory_layout: Reserved memory range 0x3ffe0000 - 0x3ffe0440
D (458) memory_layout: Reserved memory range 0x3ffe3f20 - 0x3ffe4350
D (465) memory_layout: Reserved memory range 0x40070000 - 0x40078000
D (471) memory_layout: Reserved memory range 0x40078000 - 0x40080000
D (478) memory_layout: Reserved memory range 0x40080000 - 0x40097fac
D (484) memory_layout: Building list of available memory regions:
D (490) memory_layout: Available memory region 0x3ffaff10 - 0x3ffb0000
D (497) memory_layout: Available memory region 0x3ffb6388 - 0x3ffb8000
D (503) memory_layout: Available memory region 0x3ffb9a20 - 0x3ffbdb28
D (510) memory_layout: Available memory region 0x3ffc6498 - 0x3ffc8000
D (517) memory_layout: Available memory region 0x3ffc8000 - 0x3ffca000
D (523) memory_layout: Available memory region 0x3ffca000 - 0x3ffcc000
D (530) memory_layout: Available memory region 0x3ffcc000 - 0x3ffce000
D (536) memory_layout: Available memory region 0x3ffce000 - 0x3ffd0000
D (543) memory_layout: Available memory region 0x3ffd0000 - 0x3ffd2000
D (550) memory_layout: Available memory region 0x3ffd2000 - 0x3ffd4000
D (556) memory_layout: Available memory region 0x3ffd4000 - 0x3ffd6000
D (563) memory_layout: Available memory region 0x3ffd6000 - 0x3ffd8000
D (569) memory_layout: Available memory region 0x3ffd8000 - 0x3ffda000
D (576) memory_layout: Available memory region 0x3ffda000 - 0x3ffdc000
D (583) memory_layout: Available memory region 0x3ffdc000 - 0x3ffde000
D (589) memory_layout: Available memory region 0x3ffde000 - 0x3ffe0000
D (596) memory_layout: Available memory region 0x3ffe0440 - 0x3ffe3f20
D (602) memory_layout: Available memory region 0x3ffe4350 - 0x3ffe8000
D (609) memory_layout: Available memory region 0x3ffe8000 - 0x3fff0000
D (616) memory_layout: Available memory region 0x3fff0000 - 0x3fff8000
D (622) memory_layout: Available memory region 0x3fff8000 - 0x3fffc000
D (629) memory_layout: Available memory region 0x3fffc000 - 0x40000000
D (635) memory_layout: Available memory region 0x40097fac - 0x40098000
D (642) memory_layout: Available memory region 0x40098000 - 0x4009a000
D (648) memory_layout: Available memory region 0x4009a000 - 0x4009c000
D (655) memory_layout: Available memory region 0x4009c000 - 0x4009e000
D (662) memory_layout: Available memory region 0x4009e000 - 0x400a0000
I (668) heap_init: Initializing. RAM available for dynamic allocation:
I (676) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
D (682) heap_init: New heap initialised at 0x3ffb6388
I (687) heap_init: At 3FFB6388 len 00001C78 (7 KiB): DRAM
[... 89 lines cut ...]
D (1616) efuse: coding scheme 0
D (1616) efuse: In EFUSE_BLK0__DATA1_REG is used 8 bits starting with 24 bit
D (1626) efuse: coding scheme 0
D (1626) efuse: In EFUSE_BLK0__DATA1_REG is used 8 bits starting with 16 bit
D (1636) efuse: coding scheme 0
D (1636) efuse: In EFUSE_BLK0__DATA1_REG is used 8 bits starting with 8 bit
D (1646) efuse: coding scheme 0
D (1646) efuse: In EFUSE_BLK0__DATA1_REG is used 8 bits starting with 0 bit
D (1656) efuse: coding scheme 0
D (1656) efuse: In EFUSE_BLK0__DATA2_REG is used 8 bits starting with 16 bit
D (1916) nvs: nvs_open_from_partition bt_config.conf 1
D (1916) nvs: nvs_get_str_or_blob bt_cfg_key0
D (1916) nvs: nvs_get_str_or_blob bt_cfg_key0
D (1926) nvs: nvs_close 3
D (1926) nvs: nvs_open_from_partition bt_config.conf 1
D (1926) nvs: nvs_set_blob bt_cfg_key0 216
D (1936) nvs: nvs_close 4
I (2006) BTCLIENT: Starting Central
I (2006) BTCLIENT: REG_EVT
I (2516) SETTINGS: Reading settings
D (2516) nvs: nvs_get_str_or_blob skey
I (2516) SETTINGS: Settings Read Successfully
I (2516) SETTINGS: No data differs, not writing to flash
`
It's only connected with the pc - should I see the ESP within the BT scanner app (nRF Connect) - I didn't.

dlktdr, 2022-06-23:
Yes that looks correct. I get, 
```
D (1624) nvs: nvs_close 3
D (1624) nvs: nvs_open_from_partition bt_config.conf 1
D (1628) nvs: nvs_set_blob bt_cfg_key0 216
D (1636) nvs: nvs_close 4
I (1700) BTCLIENT: Starting Central
I (1704) BTCLIENT: REG_EVT
I (2200) SETTINGS: Reading settings
D (2200) nvs: nvs_get_str_or_blob skey
I (2200) SETTINGS: Settings Read Successfully
I (2200) SETTINGS: No data differs, not writing to flash
``` 
Is what you should see when in Central Mode (Master/Bluetooth)

You won't be able to see it in the NRF app unless it's in Peripherial mode, it defaults to Central Mode (Master/Bluetooth) until the radio calls for the change. 

I just got 2.8 working well with edgeTX PR# 2072. Seems to work with this code and the PARA okay. ... Not Great yet, tho

yammaster, 2022-06-23:
yes I tooked the latest

yammaster, 2022-06-23:
YES it works now, I see the channels from the BLE in my TX!
It runs with the other board. The first one boots always twice with the brown message.
Great project, great support Cliff, thanks.

dlktdr, 2022-06-23:
Yes that's right in the current state. Have to work on this part still, it's commented out

https://github.com/dlktdr/BTWifiModule/blob/f6c66afb85341690887c44375fb37a75642e70d7/src/bt_client.c#L606

yammaster, 2022-06-23:
I used the ESPHome-Flasher as the only one flasher for the second board. It could be better when you are changing the Wiki, for easy to use and minimize mistakes.