dlktdr/BTWifiModule #10: "Invalid Header" with boards ESP32 Wroom and C3 super Mini

Thetruesaint, 2024-06-19:
Hi!, I tryed flashed release 0.9 and Compiled targets for Wroom and C3 super mini, and I get always this error:

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
invalid header: 0x69616620
invalid header: 0x69616620
invalid header: 0x69616620
invalid header: 0x69616620
invalid header: 0x69616620
invalid header: 0x69616620
invalid header: 0x69616620
ets Jul 29 2019 12:21:46

I´m sure I´m doing somethig wrong, what could be he problem?

Hillbilly-Phil, 2024-06-19:
following

Hillbilly-Phil, 2024-06-19:
Following this questions with high interest. I would like to flash the ESP32 C3 super mini as well but my knowlage of coding is 0. Willing to learn tho   Would be awesome if the super mini could be added to the supported boards. Many thanks

Thetruesaint, 2024-06-19:
Same here!, I tryed also with an ESP32 Wroom and it seems to flash ok, but I´m not able to make it work with the radio.  This is what shows in the COM Monitor:
[error esp32 wroom.txt](https://github.com/user-attachments/files/15906035/error.esp32.wroom.txt)

Hillbilly-Phil, 2024-06-20:
Ok I figured out how to flash the c3 super mini via VS Code following some youtube and the instructions by dllktdr. Coding knowlage is now at 0.1 haha   
Download the code from github as zip into download folder (NO SPACES IN THE FOLDERPATH)
Download/ Open SV Code 
Install/ Open the platformio extension

Open the unpacked zip through platformio

We need to change two things: 

1. In defines.h change the UART Pins as shown in the picture
[image]

2. In platformini.io change the COM to the COM your ESP gets recognized on (in my case it was COM15) or comment it out 
[image]

Thetruesaint, 2024-06-20:
No way mate!!.. but you said you didn´t know how to code!! haha, great!, good Job!, did you try it?, what output did you see in the COM Monitor?.. I´ve builded fw with platfomio already, but I didn´t mod any code!, what YT video did you see?.. thank you!

Hillbilly-Phil, 2024-06-20:
I first just tried flashing it like the GIT BOOK instruction said but that gave me the COM Port error and I already knew that i just have to change the UART Pins since I've read that all C3 boards are more or less the same except for the pinout...
This youTube
https://www.youtube.com/watch?v=tc3Qnf79Ny8
helped a lot to understand where to look for in all these menus.
I have no idea what the COM Monitor is sorry 
Hope you have yours running soon too! Cheers

Thetruesaint, 2024-06-21:
Well thanks to Hillibilly, we could make it work too, so, thankyou!

closed 2024-06-21 by Thetruesaint