dlktdr/BTWifiModule #11: Continuos lost trainer signal and instant recover of trainer signal

RDM3DGit, 2024-06-21:
**Radio type:** TX16S MK2 JB edition
**Version used:**
 **_- Radio FW:_** Edgetx 2.10.1 centurion cloud builded with bluetooth flag enabled
 **_- ESP32 C3 Super Mini (Modified definition.h C3Mini URTS pins):_** cloned github repository, builded it and uploaded via VSCode
**Connecting to:** Xiao BLE sense HT with 2.5.1 FW

**_Expected beheavour:_** Conection between devices stays solid and uninterrupted.

**_Actual beheavour:_** Conection stablished perfctly, but after a minute or so the conection is lost and recovered instantanously. This continues to happen every 30 o 40 seconds continusly.

Thetruesaint, 2024-06-22:
Same HW and problem, but in my case, it reconect ocasionaly and only once. This is what I see in the Monitor>

When it reconnect:
[image]

And when it lost connection:
[image]

dbpprt, 2024-06-27:
Same problem with WROOM32

RDM3DGit, 2024-06-27:
Yes in fact, i tried compiling it to a wroom at first and got the same issue, but ruse this issue pointing to the C3 mini because thats the one recommended at docs. But same issue on both boards. @dlktdr we beg for your help cliff! Pls!!!!

dlktdr, 2024-06-28:
Hi guys. I'll see if I can find some time this weekend to take a look. Are you all trying to connect to the XIAO Xiao BLE sense HT with 2.5.1 FW from Yurii's code? or also having issues with my version of HT?

RDM3DGit, 2024-06-28:
> Hi guys. I'll see if I can find some time this weekend to take a look. Are you all trying to connect to the XIAO Xiao BLE sense HT with 2.5.1 FW from Yurii's code? or also having issues with my version of HT?

Hi Cliff, thank you very much, we really appreciate your support!.

I can confirm we are all trying  Yurriis FW on xiao devices, we can't get any arduino nano 33 at this side of the world.

Thanks again!!!!

RDM3DGit, 2024-06-28:
I forgot to mention that we did connect two radios ( both tx16 mk2 with esp32 c3 with your code in it) and they didn't disconnect from each other (neither worked as trainer, but connection was solid and didn't disconnect until we intentionally disconnected thems on porpous).

RDM3DGit, 2024-06-30:
Finally I go it to work without disconnections!!!! Im actually now connected since 20 minutes ago!!!  The problem was at line 338 and 393 of bt_client.c where it was defined that gatt write requiered a response I changed them to not wait for a response and problem solved!!!
[image]

closed 2024-06-30 by RDM3DGit