dlktdr/BTWifiModule #14: Any Updates? Github actions build expired

handcraftedcc, 2025-02-11:
Are there any updates on this project?
I see there were some updated builds on github actions, but cannot download them - says they're expired.

Doncuppjr, 2025-02-11:
I noticed this too. I tried using the 0.9 release, but all I get on the console is invalid header over and over again.

Doncuppjr, 2025-02-12:
You can fork it and build it yourself. The action tags need to be updated from 2 to 4, but really just build locally and flash from linux.

handcraftedcc, 2025-02-12:
@Doncuppjr Do you know if the more recent version has resolved the binding issue, where you have to rebind the Bluetooth every time you turn off the controller/tracker?

Doncuppjr, 2025-02-12:
Not sure, I struggled with getting a damn board flashed all day. Finally, finally got it to work with Linux. I did see something in the console about writes disabled. I ordered some smaller boards that will actually fit in the radio to try tomorrow.

Doncuppjr, 2025-02-12:
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x1 (POWERON),boot:0xc (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd5820,len:0x1574
load:0x403cc710,len:0xc30
load:0x403ce710,len:0x2f58
entry 0x403cc710
I (30) boot: ESP-IDF 5.4.0 2nd stage bootloader
I (30) boot: compile time Feb 12 2025 14:49:18
I (30) boot: chip revision: v0.3
I (30) boot: efuse block revision: v1.1
I (34) boot.esp32c3: SPI Speed      : 80MHz
I (38) boot.esp32c3: SPI Mode       : DIO
I (41) boot.esp32c3: SPI Flash Size : 4MB
I (45) boot: Enabling RNG early entropy source...
I (50) boot: Partition Table:
I (52) boot: ## Label            Usage          Type ST Offset   Length
I (59) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (65) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (72) boot:  2 factory          factory app      00 00 00010000 00100000
I (78) boot: End of partition table
I (81) esp_image: segment 0: paddr=00010020 vaddr=3c0b0020 size=26ff0h (159728) map
I (114) esp_image: segment 1: paddr=00037018 vaddr=3fc93e00 size=026e0h (  9952) load
I (116) esp_image: segment 2: paddr=00039700 vaddr=40380000 size=06918h ( 26904) load
I (123) esp_image: segment 3: paddr=00040020 vaddr=42000020 size=a3094h (667796) map
I (232) esp_image: segment 4: paddr=000e30bc vaddr=40386918 size=0d3d0h ( 54224) load
I (242) esp_image: segment 5: paddr=000f0494 vaddr=50000200 size=0001ch (    28) load
I (249) boot: Loaded app from partition at offset 0x10000
I (249) boot: Disabling RNG early entropy source...
I (260) cpu_start: Unicore app
I (268) cpu_start: Pro cpu start user code
I (268) cpu_start: cpu freq: 160000000 Hz
I (268) app_init: Application information:
I (271) app_init: Project name:     BTWifiModule
I (277) app_init: App version:      v0.9-1-gfdcc334-dirty
I (283) app_init: Compile time:     Feb 12 2025 14:49:08
I (289) app_init: ELF file SHA256:  b4b849311e16d511...
I (295) app_init: ESP-IDF:          5.4.0
I (299) efuse_init: Min chip rev:     v0.3
I (304) efuse_init: Max chip rev:     v1.99
I (309) efuse_init: Chip rev:         v0.3
I (314) heap_init: Initializing. RAM available for dynamic allocation:
I (321) heap_init: At 3FC9AFD0 len 00025030 (148 KiB): RAM
I (327) heap_init: At 3FCC0000 len 0001C710 (113 KiB): Retention RAM
I (334) heap_init: At 3FCDC710 len 00002950 (10 KiB): Retention RAM
I (341) heap_init: At 5000021C len 00001DCC (7 KiB): RTCRAM
I (348) spi_flash: detected chip: generic
I (352) spi_flash: flash io: dio
I (356) sleep_gpio: Configure to isolate all GPIO pins in sleep state
I (363) sleep_gpio: Enable automatic switching of GPIO sleep configuration
I (371) coexist: coex firmware version: 49a8cdc
I (397) coexist: coexist rom version 9387209
I (398) main_task: Started on CPU0
I (398) main_task: Calling app_main()
I (398) UART: Waiting for settings to be read
I (408) SETTINGS: Reading settings
I (408) SETTINGS: Settings Read Successfully
I (408) main_task: Returned from app_main()
I (898) UART: Setting initial role
I (898) UART: Switching from mode 0 to 2
I (898) BLE_INIT: BT controller compile version [d752dea]
I (898) BLE_INIT: Bluetooth MAC: a0:76:4e:50:d2:5a
I (908) phy_init: phy_version 1180,01f2a49,Jun  4 2024,16:34:25
W (938) phy_init: saving new calibration data because of checksum failure, mode(0)
I (968) BTCLIENT: Starting Central
I (968) BTCLIENT: REG_EVT
I (1468) SETTINGS: Reading settings
I (1468) SETTINGS: Settings Read Successfully
I (1468) SETTINGS: No data differs, not writing to flash

Doncuppjr, 2025-02-12:
It maybe looks like it's going to try and save something if the settings change.......

Doncuppjr, 2025-02-16:
Cliff has merged my changes to actions, so you should have viable artifacts now. I think you can close this.

closed 2025-02-17 by handcraftedcc

Doncuppjr, 2025-02-17:
@handcraftedcc, I was talking with Cliff, and he mentioned that his HeadTracker project has a mode for the board where it acts as a receiver, and it does store the pairing.

handcraftedcc, 2025-02-17:
@Doncuppjr Oh sweet. When you mean receiver you mean receiving the trainer signal from the headtracker?

Doncuppjr, 2025-02-17:
Yes.