dlktdr/BTWifiModule #16: c3 Mini - Flashing Issues - Did I buy the wrong board?

handcraftedcc, 2025-02-19:
When using the ESPHome-Flasher everything seems to work and then I get an "invalid header"
Using 'COM4' as serial port.
Connecting...
Detecting chip type... ESP32-C3
Connecting...

Chip Info:
 - Chip Family: ESP32
 - Chip Model: ESP32-C3 (revision 4)
 - Number of Cores: 1
 - Max CPU Frequency: 80MHz
 - Has Bluetooth: NO
 - Has Embedded Flash: NO
 - Has Factory-Calibrated ADC: NO
 - MAC Address: 24:58:7C:36:17:EC
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
 - Flash Size: 4MB
 - Flash Mode: dio
 - Flash Frequency: 80MHz
Erasing flash (this may take a while)...
Chip erase completed successfully in 14.9s
Flash will be erased from 0x00001000 to 0x00004fff...
Flash will be erased from 0x00008000 to 0x00008fff...
Flash will be erased from 0x0000e000 to 0x0000ffff...
Flash will be erased from 0x00010000 to 0x000f0fff...
Compressed 15856 bytes to 10276...
Writing at 0x00001000... (100 %)Wrote 15856 bytes (10276 compressed) at 0x00001000 in 0.4 seconds (effective 357.5 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 144...
Writing at 0x00008000... (100 %)Wrote 3072 bytes (144 compressed) at 0x00008000 in 0.1 seconds (effective 358.1 kbit/s)...
Hash of data verified.
Compressed 8192 bytes to 47...
Writing at 0x0000e000... (100 %)Wrote 8192 bytes (47 compressed) at 0x0000e000 in 0.1 seconds (effective 525.4 kbit/s)...
Hash of data verified.
Compressed 918960 bytes to 460931...
Writing at 0x00010000... (3 %)

...

Using 'COM4' as serial port.
Connecting...
Detecting chip type... ESP32-C3
Connecting...

Chip Info:
 - Chip Family:  (96 %)Writing at 0x000ef4c5... (100 %)Wrote 918960 bytes (460931 compressed) at 0x00010000 in 11.2 seconds (effective 657.4 kbit/s)...

...

[22:55:49]invalid header: 0xffffffff
[22:55:49]invalid header: 0xffffffff
[22:55:49]invalid header: 0xffffffff
[22:55:49]invalid header: 0xffffffff
[22:55:49]invalid header: 0xffffffff
[22:55:49]invalid header: 0xffffffff
[22:55:49]invalid header: 0xffffffff
[22:55:49]invalid header: 0xffffffff
[22:55:49]invalid header: 0xffffffff
[22:55:49]invalid header: 0xffffffff
[22:55:49]
Serial port closed!

-------

I also tried the flash download tool, and while the flashing doesn't throw any issues, the board coms are just empty when connecting via cmd.
The LED also stopped working.

Could this be that I got the wrong c3-mini?
This is the one I ordered: https://www.amazon.com/dp/B0CS6VK6P7?ref=ppx_yo2ov_dt_b_fed_asin_title
While the listing says c3-mini the board says esp32-c3-zero on the back.

handcraftedcc, 2025-02-19:
Got the esp home flasher to complete the flash but this is what it spits out in the logs:
Using 'COM4' as serial port.
Showing logs:
[23:16:00]invalid header: 0xffffffff
[23:16:00]invalid header: 0xffffffff
[23:16:00]invalid header: 0xffffffff
[23:16:00]invalid header: 0xffffffff
[23:16:00]invalid header: 0xffffffff
[23:16:00]invalid header: 0xffffffff
[23:16:00]invalid header: 0xffffffff
[23:16:00]invalid header: 0xffffffff
[23:16:00]invalid header: 0xffffffff
[23:16:00]invalid header: 0xffffffff
[23:16:00]invalid header: 0xffffffff
[23:16:00]invalid header: 0xffffffff
[23:16:00]invalid header: 0xffffffff
[23:16:00]invalid header: 0xffffffff
[23:16:00]invalid header: 0xffffffff
[23:16:00]invalid header: 0xffffffff
[23:16:00]invalid header: 0xffffffff
[23:16:00]invalid header: 0xffffffff
[23:16:00]invalid header: 0xffffffff
[23:16:00]invalid header: 0xffffffff
[23:16:00]invalid header: 0xffffffff
[23:16:00]invalid header: 0xffffffff
[23:16:00]invalid header: 0xffffffff
[23:16:00]invalid header: 0xffffffff
[23:16:00]invalid header: 0xffffffff
[23:16:00]invalid header: 0xffffffff
[23:16:00]invalid header: 0xffffffff
[23:16:00]invalid header: 0xffffffff
[23:16:00]invalid header: 0xffffffff
[23:16:00]invalid header: 0xffffffff
[23:16:00]invalid header: 0xffffffff
[23:16:00]invalid header: 0xffffffff
[23:16:00]invalid header: 0xffffffff
[23:16:00]invalid header: 0xffffffff
[23:16:00]invalid header: 0xffffffff
[23:16:00]invalid header: 0xffffffff
[23:16:00]invalid header: 0xffffffff
[23:16:00]invalid header: 0xffffffff
[23:16:00]invalid header: 0xffffffff
[23:16:00]invalid header: 0xffffffff
[23:16:00]invalid header: 0xffffffff
[23:16:00]invalid header: 0xffffffff
[23:16:00]invalid header: 0xffffffff
[23:16:00]invalid header: 0xffffffff
[23:16:00]invalid header: 0xffffffff
[23:16:00]invalid header: 0xffffffff
[23:16:00]invalid header: 0xffffffff
[23:16:00]invalid header: 0xffffffff
[23:16:00]invalid header: 0xffffffff
[23:16:00]invalid header: 0xffffffff
[23:16:00]invalid header: 0xffffffff
[23:16:00]invalid header: 0xffffffff
[23:16:00]invalid header: 0xffffffff
[23:16:00]invalid header: 0xffffffff
[23:16:00]invalid header: 0xffffffff
[23:16:00]invalid header: 0xffffffff
[23:16:00]invalid header: 0xffffffff
[23:16:00]invalid header: 0xffffffff
[23:16:00]invalid header: 0xffffffff
[23:16:00]invalid header: 0xffffffff
[23:16:00]invalid header: 0xffffffff
[23:16:00]invalid header: 0xffffffff
[23:16:00]invalid header: 0xffffffff
[23:16:00]invalid header: 0xffffffff
[23:16:00]invalid header: 0xffffffff
[23:16:00]invalid header: 0xffffffff
[23:16:00]invalid header: 0xffffffff
[... 379 lines cut ...]
[23:16:01]invalid header: 0xffffffff
[23:16:01]invalid header: 0xffffffff
[23:16:01]invalid header: 0xffffffff
[23:16:01]invalid header: 0xffffffff
[23:16:01]invalid header: 0xffffffff
[23:16:01]invalid header: 0xffffffff
[23:16:01]invalid header: 0xffffffff
[23:16:01]invalid header: 0xffffffff
[23:16:01]invalid header: 0xffffffff
[23:16:01]invalid header: 0xffffffff
[23:16:01]invalid header: 0xffffffff
[23:16:01]invalid header: 0xffffffff
[23:16:01]invalid header: 0xffffffff
[23:16:01]invalid header: 0xffffffff
[23:16:01]invalid header: 0xffffffff
[23:16:01]invalid header: 0xffffffff
[23:16:01]invalid header: 0xffffffff
[23:16:01]invalid header: 0xffffffff
[23:16:01]invalid header: 0xffffffff
[23:16:01]invalid header: 0xffffffff
[23:16:01]invalid header: 0xffffffff
[23:16:01]invalid header: 0xffffffff
[23:16:01]invalid header: 0xffffffff
[23:16:01]
Serial port closed!

handcraftedcc, 2025-02-19:
After following what you guys did with the c3 mini super, I was able to get something working by changing uart num and pins (I modified the c3 mini super values:

#define UART_NUM UART_NUM_0
#define UART_TXPIN 28
#define UART_RXPIN 27

One of the bits I'm getting is:
␛[0;31mE (375) uart: uart_set_pin(723): tx_io_num error␛[0m
ESP_ERROR_CHECK failed: esp_err_t 0xffffffff (ESP_FAIL) at 0x420032e0
Not sure if this is an issue on the setup, or an issue with the board not wired up.
Here the full log:

ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x4038128a
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd5820,len:0x1574
load:0x403cc710,len:0xc30
load:0x403ce710,len:0x2f58
entry 0x403cc710
I (24) boot: ESP-IDF 5.4.0 2nd stage bootloader
I (24) boot: compile time Feb 19 2025 00:03:45
I (24) boot: chip revision: v0.4
I (24) boot: efuse block revision: v1.3
I (27) boot.esp32c3: SPI Speed      : 80MHz
I (31) boot.esp32c3: SPI Mode       : DIO
I (35) boot.esp32c3: SPI Flash Size : 4MB
I (39) boot: Enabling RNG early entropy source...
I (43) boot: Partition Table:
I (46) boot: ## Label            Usage          Type ST Offset   Length
I (52) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (59) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (65) boot:  2 factory          factory app      00 00 00010000 00100000
I (72) boot: End of partition table
I (75) esp_image: segment 0: paddr=00010020 vaddr=3c0b0020 size=27030h (159792) map
I (108) esp_image: segment 1: paddr=00037058 vaddr=3fc93e00 size=026e0h (  9952) load
I (110) esp_image: segment 2: paddr=00039740 vaddr=40380000 size=068d8h ( 26840) load
I (117) esp_image: segment 3: paddr=00040020 vaddr=42000020 size=a3130h (667952) map
I (226) esp_image: segment 4: paddr=000e3158 vaddr=403868d8 size=0d410h ( 54288) load
I (236) esp_image: segment 5: paddr=000f0570 vaddr=50000200 size=0001ch (    28) load
I (242) boot: Loaded app from partition at offset 0x10000
I (243) boot: Disabling RNG early entropy source...
␛[0;32mI (254) cpu_start: Unicore app␛[0m
␛[0;32mI (262) cpu_start: Pro cpu start user code␛[0m
␛[0;32mI (262) cpu_start: cpu freq: 160000000 Hz␛[0m
␛[0;32mI (262) app_init: Application information:␛[0m
␛[0;32mI (265) app_init: Project name:     BTWifiModule␛[0m
␛[0;32mI (270) app_init: App version:      v0.9-6-g967353d-dirty␛[0m
␛[0;32mI (276) app_init: Compile time:     Feb 19 2025 00:03:07␛[0m
␛[0;32mI (282) app_init: ELF file SHA256:  2cc8f26dc3ab2bba...␛[0m
␛[0;32mI (288) app_init: ESP-IDF:          5.4.0␛[0m
␛[0;32mI (293) efuse_init: Min chip rev:     v0.3␛[0m
␛[0;32mI (298) efuse_init: Max chip rev:     v1.99 ␛[0m
␛[0;32mI (303) efuse_init: Chip rev:         v0.4␛[0m
␛[0;32mI (307) heap_init: Initializing. RAM available for dynamic allocation:␛[0m
␛[0;32mI (315) heap_init: At 3FC9AFD0 len 00025030 (148 KiB): RAM␛[0m
␛[0;32mI (321) heap_init: At 3FCC0000 len 0001C710 (113 KiB): Retention RAM␛[0m
␛[0;32mI (328) heap_init: At 3FCDC710 len 00002950 (10 KiB): Retention RAM␛[0m
␛[0;32mI (335) heap_init: At 5000021C len 00001DCC (7 KiB): RTCRAM␛[0m
␛[0;32mI (342) spi_flash: detected chip: generic␛[0m
␛[0;32mI (346) spi_flash: flash io: dio␛[0m
␛[0;32mI (350) sleep_gpio: Configure to isolate all GPIO pins in sleep state␛[0m
␛[0;32mI (357) sleep_gpio: Enable automatic switching of GPIO sleep configuration␛[0m
␛[0;32mI (364) coexist: coex firmware version: 49a8cdc␛[0m
␛[0;32mI (370) coexist: coexist rom version 9387209␛[0m
␛[0;32mI (375) main_task: Started on CPU0␛[0m
␛[0;32mI (375) main_task: Calling app_main()␛[0m
␛[0;31mE (375) uart: uart_set_pin(723): tx_io_num error␛[0m
[... 28 lines cut ...]
3fc9fa70: 0x3fc9fa4c 0x00000000 0x00000002 0x3fc9ea48 0x54524155 0x00000000 0x00000000 0x00000000
3fc9fa90: 0x3fc9fa40 0x00000002 0x00000000 0x00000000 0x00000000 0x00000000 0x3fc9b610 0x3fc9b678
3fc9fab0: 0x3fc9b6e0 0x00000000 0x00000000 0x00000001 0x00000000 0x00000000 0x00000000 0x420699ca
3fc9fad0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fc9faf0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fc9fb10: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fc9fb30: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fc9fb50: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fc9fb70: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fc9fb90: 0x00000000 0x00000000 0x00000000 0x00000054 0x00000000 0x3fc9fba0 0x00000000 0x00000000
3fc9fbb0: 0x00000000 0x3fc9fbb8 0xffffffff 0x3fc9fbb8 0x3fc9fbb8 0x00000000 0x3fc9fbcc 0xffffffff
3fc9fbd0: 0x3fc9fbcc 0x3fc9fbcc 0x00000001 0x00000001 0x00000000 0x1a00ffff 0x00000000 0xcb25af9c
3fc9fbf0: 0x3fc9fb98 0x0000007c 0x00000000 0x00000000 0x3fca0950 0x00000000 0x00000000 0x00000000
3fc9fc10: 0x00000000 0x00000000 0x3fc9fc78 0x00000000 0x00000101 0x00000000 0x00000000 0x0000000a
3fc9fc30: 0x3fca0924 0x00000400 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fc9fc50: 0x00000000 0x00000000 0x3fca02f4 0x3fc9fe7c 0x3fca07c4 0x3fca076c 0x3fca08cc 0x3fca0874
3fc9fc70: 0x3fca081c 0x00000200 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fc9fc90: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fc9fcb0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fc9fcd0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fc9fcf0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fc9fd10: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fc9fd30: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000

ELF file SHA256: 2cc8f26dc3ab2bba

handcraftedcc, 2025-02-19:
Changed tx to 21 and rx to 20. Those seem to be the correct ones.
That being said I don't really get much out of the logs now. A few lines and it stops:

--- Terminal on COM4 | 115200 8-N-1
--- Available filters and text transformations: colorize, debug, default, direct, esp32_exception_decoder, hexlify, log2file, nocontrol, printable, send_on_enter, time
--- More details at https://bit.ly/pio-monitor-filters
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H
I (108) esp_image: segment 1: paddr=0003707c vaddr=3fc93e00 size␛[0;32mI (895) UART: Setting initial role␛[0m
␛[0;32mI (895) UART: Switching from mode 0 to 2␛[0m
␛[0;32mI (895) BLE_INIT: BT controller compile version [d752dea]␛[0m
␛[0;32mI (895) BLE_INIT: Bluetooth MAC: 24:58:7c:36:17:ee␛[0m
␛[0;32mI (905) phy_init: phy_version 1180,01f2a49,Jun  4 2024,16:34:25␛[0m
␛[0;32mI (965) BTCLIENT: Starting Central␛[0m
␛[0;32mI (965) BTCLIENT: REG_EVT␛[0m
␛[0;32mI (1465) SETTINGS: Reading settings␛[0m
␛[0;32mI (1465) SETTINGS: Settings Read Successfully␛[0m
␛[0;32mI (1465) SETTINGS: No data differs, not writing to flash␛[0m

Not sure if that means it's working or if there's something wrong...

dlktdr, 2025-02-21:
That last log looks correct. Once connected to a radio via the rx tx pins you specified it should connect. If you change modes on the radio. Or connect to something? You should see extra details in the log at that point.

handcraftedcc, 2025-04-02:
Was able to get this one working by changing the tx and rx pin of the supermini. Could be cool to add it as build on here as it's a great cheap option.
https://www.waveshare.com/wiki/ESP32-C3-Zero
https://www.amazon.com/dp/B0CS6VK6P7?ref_=ppx_hzsearch_conn_dt_b_fed_asin_title_2

[image]